bambulab/BambuStudio #1057: [Feature Request] Cut tool fix/enhancement

BambuLabDiscord, 2023-01-12:
This is to resubmit a request that was sent in earlier since I was told the request was too long and couldn't be processed. 

Currently when the user rotates the Cut tool plane by entering a value (such as 90 degrees) in the X axis, what BS considers the upper and lower parts are the opposite of what the user sees. When entering a value (such as 90 degrees) in the Y axis,  the two parts are side by side but the menu still asks the user to pick between the upper and the lower side.

Some suggestions: 1) label the two sides of the cut plane with a non-directional terms such as 'A' and 'B' and show the label to the user, or 2)  when user enters a positive X axis value, rotate the cut plane counterclockwise so that what the user sees is what the BS sees; when user enters a value in the Y axis, change the options to 'Left' and 'Right' parts, instead of the upper and lower parts, or 3) rotates the plate accordingly (such as via some formula) so user's view matches that of the BS.

sickerin, 2023-02-20:
Just wanted to add other enhancement suggestions to the cut tool. Prusa Slicer 2.6.0 did a pretty good job on this. https://youtu.be/ZP2_Asz7xK8?t=570

psy0rz, 2023-04-25:
closed in 1.6.0

zhimin-zeng-bambulab, 2023-10-24:
The cut tool has been improved, you can have a try in the latest version

closed 2023-10-24 by zhimin-zeng-bambulab