bambulab/BambuStudio #1058: [Issue] Error when '3D Print' a model using 3MF format from Fusion 360

BambuLabDiscord, 2023-01-12:
When using the '3D Print' option in Fusion 360 to send a model in 3MF format to BS, a new instance of BS will be opened, when the model should have been loaded onto an active plate on an already opened BS instance.  In addition, the model is placed outside of the plate boundary. Then another message pops up about BS quitting unexpectedly.

During the process, there is also a message that reads "The 3mf is not from Bambu Lab, load geometry data only", but not sure if this message is expected when loading a 3MF file from outside of BS.

SaltWei, 2023-01-13:
This is a known issue and we are discussing and handling

SaltWei, 2023-01-13:
Same cause with https://github.com/bambulab/BambuStudio/issues/980

zhimin-zeng-bambulab, 2023-01-16:
@BambuLabDiscord  Could you provide your 3mf file?  It will help us to solve the problem, thanks!

zhimin-zeng-bambulab, 2023-02-09:
Is this a crash issue, could  you describe your steps to reproduce this problem and provide your 3mf file?

zhimin-zeng-bambulab, 2023-03-23:
@BambuLabDiscord  Can you still reproduce this issue on the latest version(v1.5.0 hotfix)?

zhimin-zeng-bambulab, 2023-08-18:
The problem has not been answered for a long time, and the attempt cannot be reproduced, so close it.

closed 2023-08-18 by zhimin-zeng-bambulab