bambulab/BambuStudio #1059: LAN only mode for P1P

stewartoallen, 2023-01-12:
**Is your feature request related to a problem? Please describe.**
How do I enable LAN only mode for P1P?

**Describe the solution you'd like**
An interface option for enable LAN only mode

**Describe alternatives you've considered**
I can find no other way to send files to the P1P over the local network

**Additional context**
clearly the printer is accessible over the network. but it seems via the cloud only. I would not have purchased this printer if I had known it only worked through the cloud.

[image]
[image]

Pmant, 2023-01-12:
On the X1(C) you have to put it into LAN-Mode on the printer screen. Bambu Studio is then able to find the printer via LAN.

stewartoallen, 2023-01-12:
this is on the P1P. there is no setting on the screen for this.

Pmant, 2023-01-12:
Just wanted to point out that this feature seams to be missing on the P1P firmware not Bambu Studio. You might have more luck requesting this on the [firmware forum](https://forum.bambulab.com/c/bambu-lab-software/bambu-firmware/13) or the [feature request forum](https://forum.bambulab.com/c/bambu-lab-software/feature-requests/100). This repository is dedicated to the Bambu Studio Software.

stewartoallen, 2023-01-13:
support asked me to post here

SaltWei, 2023-01-13:
Lan mode will be added at printer firmware side in the future, but I can't promise the time for firmware.

SaltWei, 2023-01-13:
Has transferred this request to firmware team.

stewartoallen, 2023-01-25:
any way to track the request? any updates on this?

suit, 2023-02-23:
Updates on this one yet?

SaltWei, 2023-03-02:
The fiamware will come soon. Under internal test

Schokobecher, 2023-03-02:
It's out.

closed 2023-03-03 by SaltWei

SimonP1P, 2023-03-03:
Dont work verry well... Now i can turn on Lan Only but my Computer can not connect to the P1P.

[image]

stewartoallen, 2023-03-09:
Now I can "send" to the printer's SD card, but cannot see SD files remotely or start a print job from SD remotely. So not terribly useful.

Schokobecher, 2023-03-09:
> Now I can "send" to the printer's SD card, but cannot see SD files remotely or start a print job from SD remotely. So not terribly useful.

That is how the current implementation works - no matter if P1P or X1. A future update to LAN mode might change that, but is unrelated to this issue

Mycahya, 2023-03-28:
Over here on the Bambu labs forum they have shown how to use the FTP access on the X1C, so likely will work on P1P, may try later today: https://forum.bambulab.com/t/lan-only-mode/3123/15

aw-jansen, 2023-04-08:
Any updates on this? And has anyone tried the ftp solution suggested by @Mycahya?