bambulab/BambuStudio #1060: Overhang walls are incorrectly getting Bridge speeds

psiberfunk, 2023-01-12:
**Describe the bug**
Lines identified as overhanging walls are somehow getting bridge speeds applied, even though they should probably be getting the appropriate overhang wall speed applied.  This appears to be the case for overhangs above ~60 degrees at 0.2 mm layer height.  This creates the obviously wrong situation of more severe overhangs getting the FASTER bridging speed applied when it's not a bridging situation... when they SHOULD be getting the slower overhang speeds applied.

**3mf File for This Bug**
The Attached 3MF project file demonstrates this issue, where I have set all the "slow down for overhangs" speeds to 1, but the bridge speed to 122 to help illustrate the issue.  [FanScan.zip](https://github.com/bambulab/BambuStudio/files/10398256/FanScan.zip)

**To Reproduce**
Steps to reproduce the behavior:
1. Open Project file
2. Slice
3. Observe that the incorrect speed is applied to the overhanging wall.  It should be 1, but instead is 122 !

**Expected behavior**
The overhanging wall should be treated appropriately according to the "slow down for overhangs" logic.  Slicing here without Arachne.

**Screenshots**
[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows 
 - 1.4.1.52  (latest as of writing)

SaltWei, 2023-01-12:
Overhang wall is 100% floating lines, not only bridge.
And bridge speed is applied for 100% overhang. This is not bug.
By the way, if you share the model, I can give more information.

psiberfunk, 2023-01-12:
Right, but if this is 100% overhanging wall, and you have a feature called "slow down for overhangs", I reasonably expect the overhang to.. well, you know, slow down according to the overhang detection feature (at the min speed of course) !

What's the point in differentiation between bridging lines and overhanging wall lines if you're going to treat them the same ?

Wastus, 2023-01-19:
I've had this issue in my first model I made.
I shared it over on [printables](https://www.printables.com/de/model/368010-fan-adapter-120-mm-to-140-mm-with-offset). You can see the problem in picture three (the inside) where two regions are very rough.
The printer was going fast there (probably using bridge speed) instead of overhang speed. I haven't investigated as much as @psiberfunk. The model is available there, so you can check what is going on.

psiberfunk, 2023-02-17:
@SaltWei Maybe you can re-evaluate this in light of the improved slow-down-for-overhangs PrusaSlicer has in their code?  It looks like SoftFever is also pulling that in, making it work with Arachne .. beating you guys to the punch again ;)

kind3r, 2024-08-19:
Robert Cowan did a video about the issue with a workaround (sortof):
https://www.youtube.com/watch?v=HcSOz-Lsxgg

Ilovecoheed, 2026-09-02:
@SaltWei Bumping this thread as Orca has also implemented a solution for this issue, allowing for 0% overhang walls to be printed at a different speed than bridging.} Are there any intentions to add this feature to Bambu Studio?

[image]