bambulab/BambuStudio #1063: Perimeters seem to end extrusion too early.

npoulos, 2023-01-12:
It seems to me like there is a hard-coded “coasting” applied to each perimeter as the seams always leave such a huge gap between the start and end of a perimeter. 

I’ve messed with the wipe settings quite a lot to see if this gap can be wiped closed but it’s just not possible. 

When you look at the generated line in the viewer there’s a very noticeable gap where the seam is which is almost exactly what is reproduced in the print. 

Is it possible to look into if the line is actually closing a perimeter, or add a coast setting so this can be controlled manually.

npoulos, 2023-01-12:
If you paste both attached codes into https://nraynaud.github.io/webgcode/ - (removing my comments + the part after the end code in BS) you'll visually see a tiny black void at the end of the round path. I believe this is what's causing the gap.

I've attempted tests with turning off Arachne & without arcs as well (as there is arc code at the end of the BS code) and it still produces the same output.

CCS86, 2023-01-12:
I am also noticing seam gaps. 

A user configurable setting for this (like Superslicer has), would be great:

[image]

npoulos, 2023-01-12:
@CCS86 Yes! I did some searching and found that's from SuperSlicer. I found mentioned that there is a 15% gap added and extrapolated that it was put there back when LA/PA wasn't a real thing so the gap was to allow for ooze. This was likely added back in Slic3r days or whatever Prusa fork everyone started from.

I've also seen this mentioned on Prusa forums a few times and it's always swept under the rug with comments like "it's probably your retraction/wipe/LA/PA" or "it's just the visual renderer making sure you can see where the seam starts and ends" - It's 100% not. 

A fix to this would solve soooo many of the comments I see of people complaining about the seam or about the gaps they see. It's commented on so much I'm surprised no one has gone in to look yet.

CCS86, 2023-01-12:
Absolutely. It's really a holdover from pre-PA / LA times. 

You can see that I run this zeroed out on my other printer. No bulging seams.

npoulos, 2023-01-13:
[image]

In libslic3r.h

QingZhangBambu, 2023-01-16:
> 

i set this setting to 0, and compare it with the original setting. they look totally the same.
[image]

npoulos, 2023-01-16:
definitely not the same. Slow down your travel acceleration and travel speed. Acceleration to say 200 and speed to 180. Try with the 0.15 and without 0.15 at slower speeds and then you'll see how much of a difference that gap makes.

QingZhangBambu, 2023-01-16:
the upper is the loop without clipping. (LOOP_CLIPPING_LENGTH_OVER_NOZZLE_DIAMETER=0.0)
the lower is original.

QingZhangBambu, 2023-08-08:
We open the "Seam gap" parameter recently, you can edit it when you open the develop mode
[image]

SaltWei, 2023-08-23:
Support seam gap setting on BambuStudio as well. Close this ticket

closed 2023-08-23 by SaltWei