bambulab/BambuStudio #1064: [FR] Layer Time Color Scheme Display

CCS86, 2023-01-12:
Please give us a "Layer Time", or better yet "Log Layer Time" to the preview display color schemes. Superslicer has this function and it is incredibly useful.

[image]

SaltWei, 2023-01-13:
Will be added in next release which will come soon！

CCS86, 2023-01-13:
Excellent!

Having both normal and logarithmic layer times would be fine with me. But, if you lean towards only one for simplicity, please make it the log version. Look at how much more telling it is:

[image]

SaltWei, 2023-01-16:
> please make it the log version

OK

lanewei120, 2023-03-02:
already in v1.5 on github

closed 2023-03-02 by lanewei120