bambulab/BambuStudio #1065: Filenames on Printer vs Bambu Studio

DarumasLegs, 2023-01-12:
When creating a project with multiple 3mf/stl's on a build plate, I typically save this project on my computer and name it some useful name (like "Motor mount parts"), then after slicing it and hitting the "send" button to send it to the printer, the dialog presented in Bambu Studio shows that the project will be sent to the printer using the name of the project that I saved on my computer. However, after sending the file, the resulting filename on the printer is usually different than the project name. Sometimes it displays the name of the first object on the build plate, but sometimes it also just picks a name of a previously printed project randomly.

Here is a screenshot of the Bambu Studio send dialog box:
[image]

And here is a photo of the printer's LCD display showing the name of a previously printed file:
[image]

**Desktop (please complete the following information):**
 - OS: Mac OS Ventura 13.0.1
 - Bambu Studio 1.4.1.52

walterwongbbl, 2023-01-29:
Hi. In the send print dialog,  the displayed name is the file name. like  "PLACF_Gearbox_plate_1".
but printer's LCD displayed name is model name. 
if you want to change the model name.
[image]

DarumasLegs, 2023-01-29:
Thank you. That isn’t very intuitive, by the way - why not show the model
name in the send print dialog, or have the printer’s LCD displayed name as
the filename? It think it just makes more sense if the send print dialogue
shows the same name as the printer’s LCD display.

walterwongbbl, 2023-01-30:
Yes, good idel. we will synchronously display both sides as filename.

walterwongbbl, 2023-10-13:
Please download and use the latest version of Studio

closed 2023-10-13 by walterwongbbl