bambulab/BambuStudio #1066: macOS Client - Auto arrange broken when print sequence "by object" selected

Jarred-Sutherland, 2023-01-12:
**Describe the bug**
Setting the Print sequence (Others -> Special mode -> Print Sequence) to "By Object" and using the auto arrange function causes the objects to be stacked on top of one another and placed off the build plate. 

**3mf File for This Bug**
This happens with any model I have used (10-20 so far). Download a 3mf or stl. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Others "tab" -> print sequence, select "by layer"
2. Click on auto arrange, leave default values, click apply.
3. View build plate, the model will be off of the build plate.

**Expected behavior**
Models should be arranged on the build plate following the space needed for the print head to not collide with the printed objects.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: macOS
 - Version: Ventura 13.1
[image]
<img width="404" alt="Screenshot 2023-01-12 
[image]
at 5 26 42 PM" src="https://user-images.githubusercontent.com/122574442/212194408-134af7fa-6db0-4bb6-9133-5081d11e2e7c.png">

tangketan, 2023-01-13:
Did you forget attach the files?
I guess it's because your plate is Textured PEI and your filament is PLA. In previous versions we didn't allow printing PLA on Textured PEI plate because the bed temperature is higher than PLA's vitrifiction temperature. New versions have removed this logic.

Jarred-Sutherland, 2023-01-13:
I did not forget to attach, this happens with any STL/3MF. Download a random STL/3MF from printables.com and you will see this behavior, the file doesn't matter.

I am running the latest version of Bambu Studio, I received my printer yesterday and downloaded yesterday. Checking for updates results in "this is the newest version"

tangketan, 2023-01-13:
Yes, it's not right forbidding PLA on PEI plates. Previously we only have X1 series, which has closed chamber, and printing PLA on PEI plates can cause clog. Now we have P1P and we realize it's OK to print PLA on PEI, as long as the chamber is not closed. 
This issue has already been fixed. Try this version: https://github.com/bambulab/BambuStudio/releases/tag/v01.04.01.52

Jarred-Sutherland, 2023-01-13:
> Yes, it's not right forbidding PLA on PEI plates. Previously we only have X1 series, which has closed chamber, and printing PLA on PEI plates can cause clog. Now we have P1P and we realize it's OK [...]

I can confirm the beta version is fixed, thank you very much!

closed 2023-01-13 by Jarred-Sutherland