bambulab/BambuStudio #1067: [Issue] Bambu Studio crashes everytime i try to slice a large file. From Stl or Step file it just crashes.

BambuLabDiscord, 2023-01-13:
Hey just wanted to check, are y'all working on a fix for Bambu Studios? Everytime I try to slice a large step or stl file it crashes Bambu Studios. I tried smaller files with no issues but larger files crash almost every time.

SaltWei, 2023-03-02:
Please share the 3mf for us to reproduce. Thanks!

SaltWei, 2023-04-19:
Could you please share the 3mf for us to reproduce？ Thanks!

SaltWei, 2023-08-10:
Close for no response for a long time. Reopen if the ticket wake up

OriginalSyn5150, 2024-01-19:
Same issue here.
[Uploading STC-104 OM Generator Array.zip…]()

OriginalSyn5150, 2024-01-19:
> Same issue here. Uploading STC-104 OM Generator Array.zip…

Orca slicer crashes as well...

SaltWei, 2024-01-19:
Thanks for updating. Reopen.
@XunZhangBambu Please have a check on this issue

OriginalSyn5150, 2024-01-19:
Appears to be the size of the file. Another one had the same issue, but smaller file no issues.

XunZhangBambu, 2024-01-19:
> Same issue here. Uploading STC-104 OM Generator Array.zip…

@OriginalSyn5150  Hi, the link doesn't seem to be a file. Could you please share the 3mf for us to reproduce?

SaltWei, 2024-01-19:
> Same issue here. Uploading STC-104 OM Generator Array.zip…

Yes. I can not be access to the zip as well

OriginalSyn5150, 2024-01-19:
Says I cannot upload that file type.

[image]

OriginalSyn5150, 2024-01-19:
[STC-104 OM Generator Array.zip](https://github.com/bambulab/BambuStudio/files/13992885/STC-104.OM.Generator.Array.zip)

Trying again...

OriginalSyn5150, 2024-01-19:
> > Same issue here. Uploading STC-104 OM Generator Array.zip…
> 
> Yes. I can not access to the zip as well

Try again please, see above

OriginalSyn5150, 2024-01-23:
> > > Same issue here. Uploading STC-104 OM Generator Array.zip…
> > 
> > 
> > Yes. I can not access to the zip as well
> 
> Try again please, see above

Just checking in...  Any update?

XunZhangBambu, 2024-01-23:
@OriginalSyn5150  Can't reproduce in my machine. Win 10 AMD Ryzen 9 5950x
[image]

OriginalSyn5150, 2024-01-23:
> @OriginalSyn5150 Can't reproduce in my machine. Win 10 AMD Ryzen 9 5950x [...]

I am on WIN 11, 14900KS, 32GB RAM, RTX4090…. Are we thinking a driver issue perhaps?  Issue does not happen in Lychee Slicer or Prusa Slicer, just cannot output code to the A1.

XunZhangBambu, 2024-01-23:
@OriginalSyn5150  You mean you can slice it but can't output code to A1 ?

OriginalSyn5150, 2024-01-23:
> @OriginalSyn5150 You mean you can slice it but can't output code to A1 ?

No, when I click SLICE PLATE it makes it ~15-25% then crashes the entire program.

XunZhangBambu, 2024-01-23:
@OriginalSyn5150 Could you please provide the dmp file for the crash? You can refer to the following wiki link to get the dmp file.
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue

OriginalSyn5150, 2024-01-24:
> @OriginalSyn5150 Could you please provide the dmp file for the crash? You can refer to the following wiki link to get the dmp file. https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue

Here you go!
[bambu-studio.exe.34824.zip](https://github.com/bambulab/BambuStudio/files/14031507/bambu-studio.exe.34824.zip)

XunZhangBambu, 2024-01-24:
@OriginalSyn5150  Thanks!

OriginalSyn5150, 2024-01-24:
> @OriginalSyn5150 Thanks!
[image]

No problem, I tried dropping down the model into thirds, still got the message attached, so I am thinking it is related to number of triangles...

XunZhangBambu, 2024-01-24:
@OriginalSyn5150  Could you please tell me the version you are using?
[image]

OriginalSyn5150, 2024-01-24:
> @OriginalSyn5150 Could you please tell me the version you are using? [...]

Version 1.8.4.51

OriginalSyn5150, 2024-01-26:
Just checking to see if there is a fix yet...

OriginalSyn5150, 2024-01-28:
> Version 1.8.4.51

Quick update.  I installed Bambu Studio on an old laptop and have NOT had the issue.  No clue where to go from here but looking forward to hearing back.

QingZhangBambu, 2024-01-29:
@OriginalSyn5150 @BambuLabDiscord holle, maybe you can try this method, some people said it is useful. https://github.com/bambulab/BambuStudio/issues/3354

OriginalSyn5150, 2024-01-29:
> @OriginalSyn5150 @BambuLabDiscord holle, maybe you can try this method, some people said it is useful. #3354

Thank you very much, deselecting ALL CPU's and selecting them individually made it work!  You are the best.