bambulab/BambuStudio #1068: Chamber Fan auto speed based on temperature

rwoldberg, 2023-01-13:
I would like the chamber fan to select its speed based on the current temperature of the chamber and the max temperature for a given filament.   There could be a min speed so that there is a negative pressure in the chamber and fumes are exhausted through the carbon filter and then slowly ramp up based on desired chamber temp.  This would allow the chamber fan to run quite most of the time but still allow for cooling when needed.

DzzD, 2023-01-17:
+1

In addition, It could run at a minimum speed of 10% as a default speed (wich is quiet) to always get air exhausting,  and as mentionned by OP increase its speed when chamber temperature is approching the max limit fixed in °c by user setting.

This would be obviously more logical than the current state, based on bed temperature, no sens

MiscPrinter, 2023-01-21:
I would like to see this too. 

This is something I have been doing manually. The issue is the chamber temperature is not read by the gcode and allowed to be a variable which can influence the chamber fan speed. It would have to be a sub routine done outside the gcode. A command would be needed in the gcode to start this routine which the board would have to check every (insert time interval here).

uberjay, 2023-02-09:
I'm somewhat of a newbie to 3d printing, but have some experience with gcode in the CNC milling space. Apologies if I've missed something obvious here. 😅 

It would be nice if the target chamber temperature were part of the filament profile -- I've got my x1c hooked up to a fume extractor, and am putting together a system to automatically switch between recirculation vs. drawing in air from the room.

Having said that: perhaps I'm missing something -- in the BambuStudio UI a target chamber temperature is shown. I vaguely recall it being lit up when printing Bambu PC, but not while printing Generic (actually Polymaker) PETG. Is it, perhaps, just computed from other values in the profile?

DzzD, 2023-02-09:
You can change it in filamanet Gcode section using the following command :
`M106 P3 S128 ; turn off chamber cooling fan 50%`

> M106 : the command to controle FAN
> P3 : the FAN identifier (chamber), can be P1 or P2 for other fans
> S0 : one Byte power range from 0-255

uberjay, 2023-02-09:
Right, i know it doesn't have its own heater, but it seemed potentially useful to conditionalize recirculation based on target/current temperature. Thanks for the pointers to gcode refs -- I'll give it a try and see what happens. :)

GuzziRaz, 2023-11-25:
I’m impressed with most thing about my Bambu but underwhelmed when it comes to chamber fan control. You should neither set fans to off for cool plate, nor set it to 60% for a heated plate, nor hint about opening the door or lid unless actually needed.

We already have a target chamber temperature setting but it appears to only be used by X1E for heating. ALL models with a chamber fan should use the very same setting for cooling!

So what you should do regardless of plate is obviously start out with no chamber fan, then turn it on if/when needed, and continuously adjust speed between 0 and 100% as needed. Among other benefits, this reduces fan noise, which is very high on my wish list. As soon as it reaches 100% (and only then!) you could also throw the warning suggesting to open the lid - and if user does so, the fan noise will hopefully decrease over time as temperature drops allowing the printer to decrease fan speed. 

This is a trivial thing to implement. As far as I can see you don’t need to look at any other variable: Plate does not matter, filament type does not matter. The current vs. target chamber temperatures is the only parameters you need. So you can get rid of some profile G-codes with daft open loop “if PLA and Textured Plate then fan=70%” logic.