bambulab/BambuStudio #1069: Swap color in 3mf files without having to recolor or change colors in AMS

Ali229, 2023-01-13:
**Is your feature request related to a problem? Please describe.**
Should implement a way to swap colors according to your needs in the bambu slicer without having to recolor the model or change the colors physically on the AMS.

Scenario: I downloaded/received a 3MF file from a friend who might have black in slot 1 and white in slot 2, I might have these colors swapped in my AMS, now I have to either physically change the colors on AMS (and make sure to purge enough) or have to repaint and white and black parts. As you can see this would get even more complicated as more colors are involved or if the model has a lot more facets(this one in the example is simple).

**Describe the solution you'd like**
Change the slots on the Bambu slicer according to what you have on YOUR AMS, no recoloring needed.

**Additional context**
1. Screenshot 1 is what happens when I import 3mf from another user.
2. Screenshot 2 is what happens when I sync my own AMS in bambu slicer.
[image]
[image]

MadMax389, 2023-01-13:
That would be a nice feature. SimplyColor3D has a global color replace feature. The simplest workaround is to remap the colors from the print screen before sending to the printer. You can just choose the correct slot in the AMS without having to repaint.

Ali229, 2023-01-13:
A global color replace should work really nicely here.

maniusman085, 2023-01-14:
Ran into a similar issue. Would like this to be sorted out asap. It should not be hard to do for the programers.

lanewei120, 2023-03-02:
when you send to printer, you can switch the color on the ams mapping buttons
just as
[image]
please have a try

Alfiegerner, 2023-07-23:
Hi - can you please confirm what filament slicer setting is used here?

E.g. if we have blue filament selected in the slicer, and switch that colour to use green in the print settings, is assume the sliced settings persist, i.e. blue filament settings will be used.

Epoch-Hanma, 2024-08-07:
Sorry to necro this issue... Found this post because I could not figure out this issue myself.

I believe one potential problem with the way this feature is currently implemented is that the objects are sliced BEFORE the filament is selected in AMS. Files from other users will most likely have DIFFERENT slicer settings selected then whats loaded in my AMS. 

So @Alfiegerner solution is correct. Here are screenshot's workflow:

[image]
[image]

haparose, 2024-09-07:
Figured this out.  I selected the object, then clicked the paint bucket icon.  On the left side it will show the color assignments.  You can right click on that color assignment and choose "Change Filament", then select one of the other AMS options listed.  I didn't want to do it by just changing the assignment at the print screen because I suspect it will cause unnecessary filament color changes for that item, causing more waste.
[image]