bambulab/BambuStudio #1070: Slicer hang, cannot be interrupted

ahoy-jon, 2023-01-13:
**Describe the bug**
When slicing, with advanced settings (like archimedian top) on certain STL, the slicer hang. 

**3mf File for This Bug**

**To Reproduce**
[image]

When slicing this file https://www.thingiverse.com/thing:3477876/files.

**Expected behaviour**
* The slicer should not hang
* The slicer *should be able to be interrupted* (so we can go back).

**Screenshots**
[image]

**Desktop:**
 - OS: Mac
 - Version: 1.4.0.17

SaltWei, 2023-01-16:
A basic library issue from clipper and prusa also can reproduce same issue.
[image]
[image]
Will investigate further.

ahoy-jon, 2023-01-16:
Thanks a lot! if you could prioritize slicer interruption, that would be great. 

(I think we can understand the slicer cannot solve a certain path, if we could cancel the slicer, that would help in this case).

SaltWei, 2023-03-15:
Fixed. And the fix will be included in next 1.6 version

SaltWei, 2023-04-18:
[V1.6](https://github.com/bambulab/BambuStudio/releases/tag/v01.06.00.65) has been published

closed 2023-04-18 by SaltWei