bambulab/BambuStudio #1073: Bambu Studio should remember last profile used when starting

Ruiaxe, 2023-01-14:
**Is your feature request related to a problem? Please describe.**
I'm using a custom profile and not the one provided with BambuStudio. Every time I close the software and start again my profile is not selected by default. This can let to errors in prints if I forget to make the change.

**Describe the solution you'd like**
The ability to remember the last profile used and select it by default.

**Describe alternatives you've considered**
As a workaround I have to remember to make the change manually. Not ideal.

**Additional context**
https://youtu.be/TZPLeQl-8Wo

Manfredo1952, 2023-01-15:
I would setup Studio as desired and then store this setup with a name corresponding to what it reflects. It does not require any object to be placed there. You can open this stored project and Studio will come up with the stored settings, incl AMS filament settings.

morikostyles, 2023-01-19:
I just submitted a very similar feature request related to filament profiles
https://github.com/bambulab/BambuStudio/issues/1113

lanewei120, 2023-03-02:
it should be a bug not a feature
we will check it

Ruiaxe, 2023-03-02:
This issue is fixed in the latest build of Bambu Studio, thanks.

closed 2023-03-02 by Ruiaxe