bambulab/BambuStudio #1074: Auto-slice in background when parameters modified

DzzD, 2023-01-14:
Hi,

It would be very handy to have an option to enable auto-slicing when a parameter is changed as SuperSlicer do, or other slicers already do, Slicing take very low time and can be achieved in background, It can be stoppped and restart if another parameter is modified while already slicing in background.

This simple feature really miss in BambuStudio compared to SuperSlicer or other.

This could be added as an option (auto-slice in background) that can be set on or off.

akbyrd, 2023-03-25:
Big +1 to this.

I noticed there's a keybinding for slicing the current plate: Ctrl+R. This helps a lot.
[image]

FinalFrag, 2023-10-19:
Sad to see this is still not a thing :( Please add this

speedinator, 2023-11-01:
+1 Absolutely!

GuzziRaz, 2023-11-22:
I think we don't even need an option for this, as the logic is simple: If in Prepare tab, no auto slice. If in Preview tab, auto-slice upon any change.

zeropage-io, 2023-11-22:
> I think we don't even need an option for this, as the logic is simple: If in Prepare tab, no auto slice. If in Preview tab, auto-slice upon any change.

Auto slice should definitely be made optional. If you have a big model that takes quite some time to slice it is pretty annoying to have it being automatically sliced after each change.

GuzziRaz, 2023-11-22:
> Auto slice should definitely be made optional. If you have a big model that takes quite some time to slice it is pretty annoying to have it being automatically sliced after each change.

Yes, obviously. My bad!

DzzD, 2023-11-24:
> Auto slice should definitely be made optional. If you have a big model that takes quite some time to slice it is pretty annoying to have it being automatically sliced after each change.

I do agree that it is always better to be able to disable or enable such hungry feature such feature especially on poor powerfull computer, but the idea is to make it in the background only using part of the CPU/core to not reduce user experience

>I think we don't even need an option for this, as the logic is simple: If in Prepare tab, no auto slice. If in Preview tab, auto-slice upon any change.
But I  disagree with this, it is good to have it make for each modification, but once again if it does not hurt user experience, so you have an immediate response when you click on preview tab, I found this very handy on Super Slicer for years (which I do not use anymore in favor of BambuStudio), each modification stop the current slicing and restart a new one but once again as a background task only using a fraction of the full available CPU until you are on the preview tab where it then could use more CPU.

caco3, 2023-12-20:
I was using UltiMaker Cura before and really liked this feature.
I would be happy if it could get added to Bambu Studio. And I agree that it should be a low prio background job and it must be possible to disable it (resp. default=Off).

DanielLaberge, 2024-02-23:
So far this is the missing feature that irritates me the most, coming from PrusaSlicer. Please consider adding it.