bambulab/BambuStudio #1075: [FEATURE REQUEST] Tied feature request between printer and Bambu Studio for better filament management

TheCableGuy99, 2023-01-14:
**Is your feature request related to a problem? Please describe.**
I would like to be able to specify the brand and exact colour by typing the name and RGB code in on the screen if it's not one of the few available and they are then available in Bambu Studio. In reverse I would like to be able to create the brands and custom colours within the printer when loading them and they appear in Bambu Studio.

As well of this I would like some confirmation of the colour selected once you've selected it/typed in. So if you press the red colour it says it's "red" rather than just showing the colour. The reason for this is that I'm slightly colour blind and i'm not always sure i'm selecting the correct colour.

This all needs to be synced back and forth with your account (in the cloud) between the printer and Bambu Studio so that if you enter a custom brand/colour in Bambu Studio it shows up on the printer and if you type it in the printer it shows up in Bambu Studio.

**Describe the solution you'd like**
As above

**Describe alternatives you've considered**
Not applicable as it's a feature request to expand the current filament selection features

**Additional context**
Any clarification please ask.

YifanWuBambu, 2023-03-02:
We will support customer to define his/her own filament types. Besides, these types can be synchronized among device screen, bambu studio and bamby handy. It is added to our plan, although may not be included in recent releases.

TheCableGuy99, 2023-03-02:
Well I hope you do sooner rather than later because it's just a massive pain right now having to re-select your filaments each time. I look forward to this. Thanks.