bambulab/BambuStudio #1077: Allow sending of basic gcode commands for troubleshooting

joshkentYT, 2023-01-15:
**Is your feature request related to a problem? Please describe.**
I'm having some first layer issues and feel pretty helpless to troubleshoot the issue at all. I think the bed might be warped or the mesh is returning inconsistent results, but I have no way to verify this. 

**Describe the solution you'd like**
I'd like there to be an optional "Troubleshooting Mode" (or something similar) that enables the sending of gcode commands and returns the results. 

For example, in my current situation, I'd like to be able to send a `G29 V4` command and get the mesh print out returned. This way I could run it a couple times and see if the mesh is returning inconsistent results or if there is a consistent warping in the mesh. 

**Describe alternatives you've considered**
The other option would be to make the logs readable so I could at least inspect the output when a print is run.

Manfredo1952, 2023-01-15:
Did you try the manual bed tramming option (see Wiki), this means that you would level the bed with the three screws underneath the heatbed. May be the screws are lose and the bed is not leveled correctly ?

joshkentYT, 2023-01-15:
Yes, I did.

CCS86, 2023-01-16:
I think this would be a great addition.

lanewei120, 2023-03-02:
could you modify the gcode of start-gcode to have the same debug for this requirement?