bambulab/BambuStudio #1078: Dropdown menu not working

Kwintyy, 2023-01-15:
**Describe the bug**
Both dropdown menus from slicing and print plate open but you can not select the options.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
use a MacBook Pro and try exporting or sending to a already printing printer

**Expected behavior**
a clickable menu

**Screenshots**

**Desktop (please complete the following information):**
 - OS: MACOS 13.0.1

walterwongbbl, 2023-01-16:
Hi, please try to download 1.4.1 version on girhub.  version 1.4.1 fixes this issue.

Kwintyy, 2023-01-16:
Thanks, fixed in this one indeed.

walterwongbbl, 2023-01-17:
> Thanks, fixed in this one indeed.
We will continue to optimize studio.

closed 2023-01-28 by lanewei120