bambulab/BambuStudio #1081: Restore option to turn off timelapse slicer

bakaufman, 2023-01-16:
I am frustrated with always having to uncheck timelapse. Or any of the 4 checkboxes as you send a print to the cloud. It REALLY needs to retain last settings. It is maddening.

walterwongbbl, 2023-08-02:
Please update the latest studio version

closed 2023-08-02 by walterwongbbl

sonjamichelle, 2024-10-23:
Not sure WHY tis was closed. Even on version 1.9.7.52 there is no way to turn off the timelapse.

As the OP said, you ALWAYS have to uncheck the box on the LCD.

I too get tired of having to go back and clean out the timelapse videos when I forget to uncheck that blasted box!

I don't use timelapses, have no plans on using them ever. Would be nice to not have our SD card fill up with useless videos.

SD cards have a limited lifetime. Creating and deleting these useless files just racks up the read/write and shortens the life span.

psy0rz, 2024-10-23:
I would love it if they make it so all the checkboxes stay the same