bambulab/BambuStudio #1082: Please make Speed Mode a 'slider'

SwagDad99, 2023-01-16:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
It would be nice if instead of 'Silent, Normal, Sport, Ludicrous', you could just adjust a 'slider' from %50 - %166 speed.  That way, if something could print well at %133 you could do it, not just stop at %124, or over do it at %166.  You could 'fine tune' the speed you need.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

iandouglas, 2023-01-20:
I concur, I'd like to set this at, say, 75%, or 110%, or heck even slow it down to 25% in some cases.

MiscPrinter, 2023-01-21:
It is nice on my other printers to have the ability to adjust the speed mid print. Having this feature would increase the usability of the printer greatly.