bambulab/BambuStudio #1086: Display current printer on slicer screen

mugmouse, 2023-01-16:
**Is your feature request related to a problem? Please describe.**
when painting an object in a project on a new plate, the color pattern shown does not show which device it came from.  Having multiple printers with small changes in color locations this could cause a person to spend a lot of time and waste if it was shown where the AMS it was loaded from.  Thus, causing problems with inaccurately painting an object.
**Describe the solution you'd like**
Retain on the Prepare Screen the printer's name associated with the color pallet.
**Describe alternatives you've considered**
Remember to first go to device screen load the proper printer then go back to the prepare screen and download the AMS color pallet.

**Additional context**
Add any other context or screenshots about the feature request here.

bambu123, 2023-03-02:
Same as #1122

YifanWuBambu, 2023-03-02:
@mugmouse 
It is a common case that people use colors/filaments different with current installed ones (in AMS slots).
For example, there are 10 filaments in your storage room. So you add all these filaments in your project and these filaments can be used in different print plates. Before printing a specific plate, you may install its required filaments into AMS and then send this print job.

So, the filaments in the left sidebar are not bound with a specific printer.