bambulab/BambuStudio #1092: [Feature Request] Halving AMS purge & time waste with a simple trick

NairoDorian, 2023-01-17:
The idea is not mine, I just figured out such a great idea needed to be shared everywhere so the dev have a chance to see it.

https://forum.bambulab.com/t/idea-cutting-printing-time-and-filament-usage-with-ams/2974

"There is a clear weakness of the AMS system in the changing of colors, and the poop function - there should be a 3rd option in the slicer that is between print by object and by layer.

If you could print multiple layers per color (object/color/Area/layer) you would reduce AMS changes by the number of lines - 2 lines would cut things in half and would likely work on 90% of prints, and for many prints the colors are separated enough that you might not have to change filament more often than any 10 lines"

This could be game changer for this AMS purge & time waste remaining issues !

Please BambuLab notice this asap, I see great potential in this!

SaltWei, 2023-01-17:
We have notice these ams improvent!
If merge same color objects into one object, and print different color object with sequence by object, it will save a lot of time and purge.

NairoDorian, 2023-01-17:
> We have noticed these ams improvent!
> If merge same color objects into one object, and print different color object with sequence by object, it will save a lot of time and purge.

You seem to have omited the most important feature I wrote about :

It's suggesting that when printing for example 1 big object with many different colors in it, there could be an option to print many layers of the same colors without switching to the next color  for the same layers yet, thus allowing to print more than 1 layer at a time of one color and hopefully reduced the number of swap and purge by half if printing 2 layers by 2 layers
Time wasted divided by 3 if printing 3 layers by 3 layers, and so one... This will depend on the 2 clearance the nozzle give over the printed area, a few millimeters at least so a few layers, so there is potential

Do you get the idea?

Totally different than printing "per object" and also different than "per layer"

Please recognize the huge difference there and the amazing potential, don't skip this thinking this is already possible. It's not currently possible and you should definitely try to incorporate this if possible :)

NairoDorian, 2023-01-18:
"The print by object is what is shown in the video, but if you print one object (or multiple) with color changes then you change color for each layer, whereas you should be able to get away with printing multiple layers between color shifts by zoning the print."

OmiCron07, 2023-01-24:
I think this is possible because this is already an option for the infill: infill combination. In a two days print, I saved 3-4 hours. This print has 1066 colour changes and these changes happened almost at all the layers.

ewoolsey, 2023-05-01:
+1 on this. This feature would be an absolute **game changer**. IMO this feature should be top priority.