bambulab/BambuStudio #1093: Add Creality Ender 5 as an additional option

JoeMakerDude, 2023-01-17:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

SaltWei, 2023-01-17:
We don't have this printer. It would be great if you can share presets for Creality Ender 5.

SaltWei, 2023-01-17:
And we can add the presets into Bambu Studio

coolshrimp, 2023-02-06:
I came here to ask/offer I have multiple Ender 5 Plus that iv been creating a config file for and testing i can share it to be included, i use octoprint and was going to request another feature, since we can send a print to octopi API and it works great it would be nice to be able to view and control the printer in the device tab.

SaltWei, 2023-07-11:
Included in https://github.com/bambulab/BambuStudio/releases/tag/v01.07.00.53
[image]

The initial preset is from OrcaSlicer. Thanks!

closed 2023-07-11 by SaltWei