bambulab/BambuStudio #1094: Add ability to send gcode to MoonRaker Fluidd interface for remote printing on Creality Printers

JoeMakerDude, 2023-01-17:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

SaltWei, 2023-01-17:
[image]
I have no creality printers. But I have voron with Fluidd.
And I can directly send gcode to voron.

SaltWei, 2023-01-17:
I think it should work for Creality Printers as well

cotepat, 2023-01-17:
Even better... use the SoftFever branch which is optimized for Klipper. Fluidd or Mainsail is directly integrated in device Tab. 

https://github.com/SoftFever/BambuStudio-SoftFever