bambulab/BambuStudio #1095: [Feature Request]Multiple Lines for Text Shape & More

rhys100, 2023-01-17:
Hi, I have the following feature requests for text shapes

1. Allow Multiple lines in the text creation
2. Auto Wrap, shrink size to text shape. EG if you have a rectangle, then when you create text you should be able to size the rectangle to the rectangle and the text will be 2-3 lines on it.
3. Allow text to be moved and not auto bind to object. (if you make one text, then create a 2nd text, you can then no longer move text 1 around as it stuck to the object.

lanewei120, 2023-03-02:
could you have a try for the new version v1.5

for 3. you can move the text in the text tool around the object and also can move it not bound to object outside of the tool

rhys100, 2023-03-08:
@lanewei120 its great! i think the last thing to improve would be letting user specify how tall the text is. For my designs we usually only want half of what bambu is doing so we are still doing the text in fusion at this point because of that issue

zhimin-zeng-bambulab, 2023-08-18:
https://github.com/bambulab/BambuStudio/issues/2086

mikemsd, 2024-02-13:
Would like to add my support for this feature as well. I am trying to make personalized luggage tags, and it's tricky lining up multiple lines. It would be way easier to type everything out and then place it in a block.

GilchristT, 2025-04-23:
> Would like to add my support for this feature as well. I am trying to make personalized luggage tags, and it's tricky lining up multiple lines. It would be way easier to type everything out and then place it in a block.

+1 for this request

randfee, 2026-01-18:
oh my... it is ridiculous that the text tool can't do line breaks!!!
Now I'm stuck having to look into developer mode so I can  use OrcaSlicer with my new Bambulab machine :-(

randfee, 2026-01-19:
I don’t think I can downgrade my the H2C

Tricoms1, 2026-02-14:
This request has been around for three years without a fix. Does anyone on the dev team actually read these feature requests?

etkaar, 2026-09-13:
Upvote!

BenTristem, 2026-09-13:
I would like to see this

Haidiye00, 2026-09-22:
https://github.com/bambulab/BambuStudio/releases/tag/v02.08.04.57
Hello, please update to the latest version. The latest version is already supported. Thank you for your support. If you have any further questions, please feel free to raise them. We will continue to communicate and thank you again

closed 2026-09-22 by Haidiye00