bambulab/BambuStudio #1096: Automatic Filament switching on Runout detection uses wrong slot.

randyre, 2023-01-17:
**Describe the bug**
When filament in a slot runs out AMS switches to a like filament, but not the same filament.

**To Reproduce**
**Steps to reproduce the behavior:**
1. Load two of the same color filament but not next to each other. Then load two more of the same filament, but in different colors.
2. Start printing with one of the same colored slots and allow it to runout
3. Filament switches to the slot next to the runout filament

**Expected behavior**
Should have switched to the filament of the same color if it was available regardless of what slot that filament was in.

StoneLiBambulab, 2023-01-17:
Hi @randyre, 
Let me to confirm the behaviour of printer, not a bug of studio.

Thanks
Stone Li

StoneLiBambulab, 2023-01-17:
Hi @randyre ,
This is a known issue of printer.  It will be fixed in next release of printer firmware, it will be released early next month.

Thanks!

Stone Li

randyre, 2023-01-18:
@StoneLiBambulab   thanks for the fast responce and confirmation. 👍

michalss, 2023-01-29:
same here i hope this will be fixed since its giving me hard time.. :(

StoneLiBambulab, 2023-10-24:
Close this issue as fixed, open a new one if there is any problem.

Thanks!

Stone Li

closed 2023-10-24 by StoneLiBambulab