bambulab/BambuStudio #1098: Add ability to handle post-processing scripts

johnmbaughman, 2023-01-17:
**Is your feature request related to a problem? Please describe.**
To enhance the output handling of a gcode file, the application needs a Post-processing script option similar to PrusaSlicer. This will help make tasks such as filament calibration easier, faster, and more consistent.

**Describe the solution you'd like**
As part of advanced or expert options, there needs to be an option to plug in a script handler that will modify the generated code file in place.

**Describe alternatives you've considered**
Handling the gcode file manually after slicing and exporting the gcode file.

**Additional context**
In PrusaSlicer this option is located here under the Print Settings tab.
[image]

lanewei120, 2023-03-02:
could you give more details for how to use the postscripts to improve "filament calibration with temperature and retraction towers"

johnmbaughman, 2023-03-02:
Some implementation of what PrusaSlicer does. 

https://help.prusa3d.com/article/post-processing-scripts_283913

I currently have a couple of models that are dynamic in size depending on the settings used to generate them in OpenSCAD that are used for retraction and temperature calibration. I use a formatted string along with the layer number inserted during the slicing/exporting of the gcode file. This file is then passed into a .Net application that inserts necessary changes for the calibration (i.e. temperature or speed changes) based on the formatted string.

Being able to continue using this feature would make this process easier as I don't have to manually run the application after exporting the file.

-John

SaltWei, 2023-04-18:
Has been included in v1.6 public beta

closed 2023-04-18 by SaltWei