bambulab/BambuStudio #1099: [Feature Request] Improved chamber fan control

BambuLabDiscord, 2023-01-17:
You added the ability to control the chamber fan from within Bambu Studio (and Handy, printer's touch screen) but the G-code overrides any changes.
Could you add a 'G-code' option to the current On/Off options?
On - permanently on (overrides G-code)
Off - Permanently off (overrides G-code)
G-code - Returns to G-code control

Kwintyy, 2023-01-17:
It would be handy if you can set the chamber fan speed per filament or material. Even better if its speed would be automatically regulated according to the set chamber temperature.

suit, 2023-03-02:
Also please consider adding the Chamber Fan Control for the P1P - it can be currently only controlled via GCode but there is no option to control it from Bambu Studio or the Printer itself.

For Bambu Studio when you switch to the controls, you see the Chamber Fan briefly, but then the option got taken away.

Since you can buy the fan from the shop and plug it in and it is also working properly, it would be useful if you build your own enclosure do make use of the fan just like the X1 does

mcattle, 2023-06-06:
I would also like to see the option of automatically adjusting the chamber fan to the chamber temperature, as well as a configurable maximum chamber fan speed setting (default to 100%).  I personally set it to 80% for hot bed PLA prints, but currently the G-code will override it back to 100% on every filament change, which is rather loud in my environment.