bambulab/BambuStudio #1100: Spiral Z-hop causes flexible blobs.. no way to switch to classic and fix them.

psiberfunk, 2023-01-17:
The spiral Z-hop the slicer implements is generally great... except for flexible filaments.. where it often makes nasty external artifacts (see attached picture).  This can be fixed by switching back to the "classic" z-hop (verified using the SoftFever Fork).  Bambu should fix this defect by allowing the user to make per-filament overrides to the Z-hop type, and set the default TPU profile to "Classic" behavior, so that everyone can have improved TPU prints.

[image]

YifanWuBambu, 2023-02-09:
We will make the lift type an option and may be different for filament types.

SaltWei, 2023-03-01:
[image]
Available in v1.5.0 Public beta

closed 2023-03-01 by SaltWei

SaltWei, 2023-03-10:
> Bambu should fix this defect by allowing the user to make per-filament overrides to the Z-hop type, and set the default TPU profile to "Classic" behavior, so that everyone can have improved TPU prints.

Can be set in filament setting as well. But we recommand using "Slope" for TPU.

psiberfunk, 2023-03-10:
@SaltWei  is slope's behavior described anywhere ?

SaltWei, 2023-03-13:
> @SaltWei is slope's behavior described anywhere ?

Just make the spairl lift to be straight slope but with same angle.

psiberfunk, 2023-03-13:
And why is this better than traditional for TPU?