bambulab/BambuStudio #1101: First Layer Inspection: Should be Toggled Option in BS, not Global Setting on Printer

CCS86, 2023-01-17:
I believe that first layer inspection should be one of the toggled options when starting a print via BS or the printer controller. 

Depending on the nature of the print and whether you will be present during the first layer, you will change whether you need the printer to inspect the first layer. It's not something best suited to being tucked away in a global printer setting.

CCS86, 2023-12-14:
@lanewei120 Any updates?

lanewei120, 2023-12-15:
Hi

thanks for your suggestion

[image]
currently this is a global setting which can be set on BambuStudio

as this option is not a frequent used option, we just put it in above position,
Also we will monitor customers' feedback and discuss whether to move it before start printing