bambulab/BambuStudio #1104: Can't reprint from History to specific printer

bolsoncerrado, 2023-01-17:
**Describe the bug**
Can't reprint from History to specific printer On a multiple printer setup.

**3mf File for This Bug**
Happens with any file. 

**To Reproduce**
Unsure fully how to reproduce. 

I open the Handy app on Android OS. One of 3 printers available. I click on History, a print job, then try to Reprint but I can't choose the available printer. Instead I get a busy printer listed and no option to choose any other. 

**Expected behavior**
An actual list of available printers to choose a reprint to. 

**Screenshots**

[image]
[image]

**Desktop (please complete the following information):**
 - OS: Android OS 
 - Version latest

bolsoncerrado, 2023-01-17:
First screenshot shows alpha and beta printing, while gamma is available and idle.

Second screenshot shows i try to resend/reprint a file from History menu, but gamma is not listed, while beta is busy it is shown as available. 

Originally this print job was sent to beta thru the cloud. 
I was able to send it to alpha somehow, but unable to send it to gamma.

I tried to unlink and relink gamma to the cloud account but no change.

SaltWei, 2023-01-18:
This is not slicer issue. Please trun to https://github.com/bambulab/Bambu-Handy to report app issue.

bolsoncerrado, 2023-01-18:
Sorry you're right!

bolsoncerrado, 2023-01-18:
Reported

closed 2023-01-18 by bolsoncerrado

SaltWei, 2023-01-18:
> Reported

Thanks! I also had transmitted this tickets to app team internally.