bambulab/BambuStudio #1105: Timeplapse video unusable with head parked in last frame.

randyre, 2023-01-18:
**Describe the bug**
Video recording in traditional timelapse still has the print head in the way on the last frame.

**Expected behavior**
The Head should park and one last frame produced to leave the object in view.

**Screenshots**
[If applicable, add screenshots to help explain your problem.](https://user-images.githubusercontent.com/60475095/213039149-15e862a8-9494-4861-8958-abed8083a9f7.mp4)

Workaround is to edit the video with editing software, but really should not be needed is the timelapse does it job correctly.

sodachen22, 2023-01-18:
In traditional timelapse mode, the timelapse recording does not care about the poisition of the tool head, so the tool head is not parked in the last frame. Yes, it shall be optimized to keep the last frame is a whole picture of the printer object.

randyre, 2023-01-19:
@sodachen22   Thanks, yeah, just the last frame needs to be optimized and park the head so the entire build plate can be seen, all the others its fine if the head is in the way, as that's the tradeoff of the speed with traditional timelapse.

Thanks
Randy

tangketan, 2023-04-18:
We have improved traditional timelapse, so the toolhead will park at the excess chute in the last frame.

randyre, 2023-08-29:
Works. Thanks. Will close this.

closed 2023-08-29 by randyre