bambulab/BambuStudio #1107: Print Plate in BS Overwrites Files in Internal Storage

CCS86, 2023-01-18:
Even when the new job has a unique name, initiating a print from BS overwrites the last user job in the internal memory.

Why are sample files preserved and actual user jobs overwritten? Why can we only have one user job stored on the printer for fast access?

lanewei120, 2023-01-20:
Hello

thanks for reporting this bug

which printer are you using?X1/X1C or P1P
and which studio version and printer firmware are you using?

and did you have a sdcard plugged in when printing?

CCS86, 2023-01-20:
It is an X1C

Yes, I have an SD card installed for time lapses.

lanewei120, 2023-01-21:
the internal memory only saves the current print job
and the history jobs will be saved in sdcard
we can find them on printer and print again from printer

is this what you want?

CCS86, 2023-01-24:
Is there a reason that history jobs cannot be stored on the internal memory?

Storing on the SD card would be okay, but that is not happening. 

Maybe the fact that my primary computer is on our LAN and the X1C is on our WIFI network, and these are on different subnets, is causing files not to be stored correctly on the SD card? I'm still not sure I understand that network limitation.

When you send a print job the printer itself should move the last print job to the SD card.

lanewei120, 2023-01-25:
@sodachen22 Could you give more explanations on this logic on printer side?

sodachen22, 2023-01-27:
@CCS86, In the latest firmware, the 3MF for the printed job is in SD card. And there is only one history printed job through cloud is kept in internal memory.

CCS86, 2023-01-27:
> @CCS86, In the latest firmware, the 3MF for the printed job is in SD card. And there is only one history printed job through cloud is kept in internal memory.

It is not though. 

The only print file on my SD card is a gcode file I manually copied there via computer. I have sent many jobs via BS and none appear on the SD card.

sodachen22, 2023-01-28:
@CCS86 Are your firmware are latest and the SD card have enough free space?

CCS86, 2023-01-28:
> @CCS86 Are your firmware are latest and the SD card have enough free space?

Yes.

sodachen22, 2023-01-31:
The lastest firmwre is still in beta testing.

CCS86, 2023-02-09:
Before the 1.04 FW release, my printer started storing **some** print jobs to the SD card. But for some reason, most of them are not being retained.

Here is my SD card contents:

[image]

But my print job history has over 40 prints in it.

Even since updating to 1.04, the behavior has not changed and my most recent job has not been moved to the SD card. It is still sitting in the internal storage, with the wrong name:

[image]

mizone0426, 2023-02-10:
Is there any history print in "Print Cache" sheet?

CCS86, 2023-02-10:
> Is there any history print in "Print Cache" sheet?

Not for me. It is always empty.

I think the only time jobs go there is if it is already printing and you send a new job.

sodachen22, 2023-02-11:
@CCS86  There is an option for print cache here, is it enabled? Is there cache directory created in SD card?
[image]

sodachen22, 2023-02-11:
> I think the only time jobs go there is if it is already printing and you send a new job.
Yes, this function is caching 3MF of the printed job after "Cache cloud print file to MicroSD card" is enabled. It is not syncing print history of from the cloud service.

CCS86, 2023-02-11:
> > Is there any history print in "Print Cache" sheet?
> 
> Not for me. It is always empty.
> 
> I think the only time jobs go there is if it is already printing and you send a new job.

I'm not near the printer right now, but in Bambu Handy and Bambu Studio, I do not have that option. Both end with "auto-recovery from step loss"

sodachen22, 2023-02-11:
> I'm not near the printer right now, but in Bambu Handy and Bambu Studio, I do not have that option. Both end with "auto-recovery from step loss"
The option is only on the screen of the printer now. One of the reasons is Handy/Studio does not support browsing and printing 3MF on the SD card of the printer remotely.

CCS86, 2023-03-01:
Even with that option enabled, my print jobs are not reliably accumulating on the SD card.

sodachen22, 2023-03-03:
The total cache file number and size are limited. Would you please take a look at the description of this feature in the release notes?
https://wiki.bambulab.com/en/x1/manual/X1-X1C-AMS-firmware-release-history
[image]

CCS86, 2023-03-03:
That's fine if you want to limit the "print cache" to 8 files, but why can we not have an option to keep a running history of **all** prints on the SD card, if we have plenty of space to do so?