bambulab/BambuStudio #1110: Chamber Fan Setting during Slice is not effeting the print

the0bone, 2023-01-19:
**Describe the bug**
I slice a model and set the value for the Chamber Fan to 0.
But it is not executed during the print. During startup process it is 70% and for the print it starts go to up to 100%.

**3mf File for This Bug**
Independent of any file.

**To Reproduce**
Edit Filament Settings -> cooling -> last item for additional_cooling_fan_spped

**Expected behavior**
Set at least the speed to the value that is entered After the startup procedure.

**Screenshots**
[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version Bambu Studio 1.4.1.52

the0bone, 2023-01-19:
ok, it makes sense to call the fan below the Chamber Fan the Aux Fan...
[image]

closed 2023-01-19 by the0bone

LuxemGamer, 2023-01-29:
You have to go to Filament Start G-Code and edit the value S255 to something between 0-255.

It checks if the bed temperature is above 45C and sets the speed of the chamber fan that way.

The Aux fan and the part cooling fan can be set in the filament settings.

[image]