bambulab/BambuStudio #1111: Updated to new version lost profiles MACos

TheHogarthWorkshop, 2023-01-19:
*Updated to new version lost profiles MACos

lanewei120, 2023-01-20:
@TheHogarthWorkshop 
1. have you used the cloud-sync for your user profiles before?
and this time do you use the cloud-sync?
2. could you give some pictures of the files in /Users/user.name/Library/Application Support/BambuStudio/user/default
and /Users/user.name/Library/Application Support/BambuStudio/user/user_id_number

MiscPrinter, 2023-01-21:
This can sometime be fixed by rebooting. 

To prevent this from happening again, as it has happened to me a few times, create a backup of the data by doing the following. 

open finder and go to the main hdd, cmd+sht+. to show the hidden files, users->[user name]->Library->Application Support->BambuStudios->user->[user id numbered folder] In that folder is all your custom profiles. Copy that folder and save it somewhere. Next time BS fails to load them, quit, drop that folder in and re-open the program. 

When that folder is empty the software has nothing to load and moves on.

lanewei120, 2023-01-22:
I am sorry for this issue

we have added some refine logic to fix a possible issue which will delete user presets wrongly
however, the version 1.4.2 haven't include it

I guess you don't use clound-sync, so the presets only save in local
before next version(1.5) out, please just use this way to backup the presets

thanks

TheHogarthWorkshop, 2023-01-22:
Okay, now on cloud-sync

SaltWei, 2023-03-02:
1.5.0 Public beta has been released on github

closed 2023-03-02 by SaltWei