bambulab/BambuStudio #1112: Filament calibration dialog combobox is empty when using only external filament

asierralozano, 2023-01-19:
**Describe the bug**
I have a P1P printer WITHOUT the AMS unit. If I try to follow the steps decribed [here](https://wiki.bambulab.com/en/software/bambu-studio/extrusion-calibration), using the external filament is supported.

When I click on the `cali` button, the filament combobox is not listing any filament.

[image]

**3mf File for This Bug**

If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Follow the Calibration guide on the wiki

**Expected behavior**
Have filaments to choose

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

Noskowitch, 2023-01-19:
I have the same problem with this feature. I completely reinstalled bambu studio but it didn't help.

asierralozano, 2023-01-19:
> I have the same problem with this feature. I completely reinstalled bambu studio but it didn't help.

Yep. I've tried reinstalling it, and even removing the local data folder, and same result, still no filament.

StoneLiBambulab, 2023-01-20:
Hi @asierralozano ,
I  reproduce the issue. and found the nozzle value is convert to wrong value in French.

[image]

I will change another way to do this convertion to avoid this issue.

Please change to language English to do extrusion calibration currently.

Thanks!

Stone Li

StoneLiBambulab, 2023-01-20:
Hi,
    For the expression for 0.4 is "0,4" in french, so the convertion is failed. i will fixed it.

Thanks！

Stone Li

SaltWei, 2023-03-01:
FIx is available in v1.5.0 Public Beta

closed 2023-03-01 by SaltWei