bambulab/BambuStudio #1115: Improve Ironing

DzzD, 2023-01-19:
Ironing is one of my favorite feature, unfortunately I found that in Bambu Studio it is quite inconsistent and need to be tuned by filament.

On my other printer (CR10s Pro), I managed to get consistent Ironing with SuperSlicer.

**Would be nice to :**
- let overide ironing settings on filament settings
and/or
- review how Ironing is done on BambuStudio (I suppose something is wrong but cant say exactly what it is, I noticed that the Ironing pass use 0° on the 45° top face filling, using 90°/perpendicular may help)

DzzD, 2023-01-20:
Contrary to what I suggested  & according to [this article from Prusa](https://blog.prusa3d.com/make-top-surfaces-super-smooth-ironing-prusaslicer-2-3-beta_41506/),  45° is not bad and should give best results.

I have done some more and more and more and more tests, I found that probably the ironing pass is not good with some filament because of **overheating, wich is due to the slowness of the nozzle and the high temperature requiered to print at high speed**.

I managed to find a sweet spot wich works well for most of my filament. Mainly, it is a matter of reducing printing filament temperature that are sensible to overheating and Ironing parameters does not need to be modified and can be shared by all filaments.

The downside is that make impossible to print with the highest filament temperature with Ironing enabled and so it may requiere to slow down print (or filament max volumetric speed)

The original request is still valuable, but maybe only one additional filament setting could do the job "ironing nozzle temperature"

Another way would be to add a new gcode placeholder **"is_current_layer_ironed"**, it would make easy to lower temperature for ironing layers.

Last and **best solution IMHO** solution would be to add a new ironing setting **"Ironing Nozzle temperature"**

**Side note :** I really hopes that we will get CHT Nozzle on Bambu Lab printers, this will make the filament temperature more consistent and better transmitted to filament. No more need to set very high temperatur to print fast (and obviously, possibility to print with a high flow : 0.6 nozzle, 0.3 layers)

SaltWei, 2023-04-18:
@QingZhangBambu Please evaluate this feature request during next sprint

QingZhangBambu, 2023-04-23:
Good point, i will do some tests

BambooBugFixer, 2023-07-31:
I printed several cube models with different angles between the ironing and the top infill (0°, 45°, 90° separately) with a layer height at 0.16 mm, and I found no significant difference among the ironing of these models.

SaltWei, 2023-08-03:
> I printed several cube models with different angles between the ironing and the top infill (0°, 45°, 90° separately) with a layer height at 0.16 mm, and I found no significant difference among the ironing of these models.

Please share pictures！！！

SaltWei, 2023-08-03:
What printer are you using？what filament are you using？Please provided full information about your test @BambooBugFixer Otherwise how user trusts your test?

BambooBugFixer, 2023-08-03:
Thanks for your kind reminding. I used P1S and I tried tow filaments: ABS and PLA marble. There is no significant difference for the ironing on both filaments. Here is a typical example. 
[image]
[image]

angelorodem, 2026-10-10:
Bumping this feature request, as it's still highly relevant.

Ironing speed, flow, and line spacing are heavily dependent on the filament being used. Having these settings tied exclusively to the Process profile means we need to manually adjust them every time we switch materials, even when using the AMS.

Ideally, we should be able to save ironing parameters directly in each filament preset, with optional overrides of the Process settings.

OrcaSlicer already supports per-filament ironing overrides, so it would be great to see feature parity in Bambu Studio.

Any updates on whether this is planned? This would be a significant quality-of-life improvement, especially for AMS users.