bambulab/BambuStudio #1120: Improve readiability of the humidity information in the AMS panel on the Device page

davidjericho, 2023-01-19:
**Is your feature request related to a problem? Please describe.**
I use 4k monitors, and the icon for the AMS humidity display is quite small even on a 32" monitor. If I focus on it, I am able to figure it out, but the bars and the colours are difficult to make out at a glance.

**Describe the solution you'd like**
I propose a few ideas could be considered:
1) Increase the size of the icon so it's larger than 4x5mm on my 32" screen, possibly using screen DPI information if possible.
2) Change the Orange for high to yellow to better distinguish it from red.
3) Add textural presentation of the values, such as "Humidity: High"
4) Change the desiccant bars to a percentage display or a gauge as it used elsewhere, maybe a popout similar to the fan speed icons.

TheCableGuy99, 2023-01-20:
A popup alert that the humidity is low might also be an idea. But with the option to dismiss the alert for a day, week or month?

Merlwynd, 2023-01-25:
Yes! +1….

TheCableGuy99, 2023-01-25:
> And for the love of God, stop popping up the dialog on mouseover.

lol

Ali229, 2023-02-20:
It is hard to tell which which bar is filled and which isn't due to the color choices, this size needs to increase and the colors need to be different.

psy0rz, 2023-04-25:
check out https://github.com/bambulab/BambuStudio/releases/tag/v01.06.00.65

can probably be closed now.

TheCableGuy99, 2023-04-25:
Still looks tiny and could be miles better if you ask me! Not much of an improvement!

davidjericho, 2023-04-25:
Looking at the release notes, I wouldn't call that any more readable. Taking the image from the release notes, it hasn't changed substantially. 
[image]

walterwongbbl, 2023-04-27:
Hi.
I will provide feedback to our designer regarding the humidity icon

gitX14N, 2023-07-03:
I also commented on #1614... I didn't realize there was an older request for this "feature" ask. So forgive the duplication...

My eyesight isn't what it once was, and my printer room has an average humidity in the mid 60s. Staying on top of desiccant in the AMS is important. The icon in both the printer display is ridiculously small. I would say the slicer too... but I'm really not sure if the enlarged popup is showing the actual status, or if that is a static image? I assumed the latter. For improved usability and intuitiveness, I think it would be much better to display the humidity data outside of the crammed filament UI in the slicer. Push the "Tips" section somewhere else maybe, and use that UI real estate for the humidity info and desiccant status.

In #1614, this feature was slated as being close to release on Apr 16th. It's July... any chance you can bump this into the next build? For someone that has to constantly keep an eye on humidity, this is helpful. Thanks all.

kriordan37, 2023-07-19:
A guage that displays the numerical value would be nice. also as stated in a previous issue. it would be nice to have a clean mqtt topic

closed 2023-10-13 by walterwongbbl