bambulab/BambuStudio #1121: Unacceptable quality for the price

user897943, 2023-01-19:
**Describe the bug**
A clear and concise description of what the bug is.
Attaching a sample piece I printed earlier - the name and address on the tag are fake for demonstration purposes.
Just look at those gaps. Those layer lines.

I believe the latest firmware for the X1C as of today is worse than the firmware we had before Xmas 2022. The retraction seems much worse leaving horrendous gaps. Either firmware or the slicer which to be honest I'm not sure as both printer and slicer were updated at the same time.

I bought this for business, I need it to be top quality, please sort it Bambu Lab, it's not a cheap toy, you're selling it as a premium printer so make it behave so. Thanks.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
load file, slicer settings as follows:
bambu lab 0.4 nozzle (carbon)
cool plate
generic pla
0.16 optimal @bbl x1c

quality: all default but change to arachne
wall tran thresh angle 59
wall tran filter margin 100
wall distribution count 3
min wall width 100
min feat size 1

Others:
disable prime tower

See screnshot for preview.

See pics of finished part.

NOT GOOD!

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]
[image]
[image]

[image]

theogott, 2023-01-23:
You should attache the 3mf file so people can find out whats wrong. Possibly its a issue with your settings.

SaltWei, 2023-01-28:
Please share 3MF file for us to reproduce