bambulab/BambuStudio #1122: [Feature Request] Better filament management

TheCableGuy99, 2023-01-20:
**Is your feature request related to a problem? Please describe.**
When adding filament the only options are to add Bambu Lab filaments and "Generic". This is a problem because we need to be able to identify EXACTLY which filament is located in which slot and which manufacturer.  

The reason we need this rather than just selecting "Generic" is because different filaments from different brands have different flow rates, different volumetric flow rates and different settings. Some people like to go to the trouble of calibrating their filaments as using a generic profile is not enough. I know I and many others do this.

Currently when selecting "Generic" we have to note down exactly which brand and type of filament we have put in there because "Generic" doesn't identify it and then we need to select the profile created for that filament in the slicer and the correct slot. 

It's currently a nightmare as I have 7 AMS units and i'm ordering more soon and I have to note down and document exactly what is in each slot and then each time i'm slicing I have to re-enter it for each printer as it just goes back to "Generic" which is the wrong profile for that filament.

**Describe the solution you'd like**
We need to be able to add filaments on the printer and name the brand with a popup keyboard on the LCD display. If we have created a profile in the slicer for a specific filament, that must be available to select at the printer. And vice-versa... if we create a filament brand in the printer we need to be able to select it in the slicer. The two need to sync-between each other and it needs to remember it.

**Describe alternatives you've considered**
Writing it down and documenting each slot is the only way to resolve this currently and it's a MASSIVE pain!

**Additional context**
Add any other context or screenshots about the feature request here.

SaltWei, 2023-03-01:
Available in v1.5.0 Public beta

closed 2023-03-01 by SaltWei

TheCableGuy99, 2023-03-01:
> Available in v1.5.0 Public beta

I've just tested this and it doesn't appear to be working...

I see the is a new button to "reSync". I presume "Resync" pulls everything from the printer and overwrites the software? If so I'm not sure exactly what "Sync" does?

What I do know is if I set all the filaments, close the app and re-open it the filaments are the same... great so far...

But then if I switch to another printer under the "Device" tab and set the filaments for that printer and then move back to the other printer it's showing the filaments for whatever printer I set last. It's not setting the filaments and keeping them when switching between printers meaning I have to keep setting the filaments every time I switch printers.

lanewei120, 2023-03-02:
> nother printer under the "Device" tab and set the filaments for that printer and then move back to the other printer it's showing the filaments for whatever printer I set last. It's not setting the filaments an

after switch back to the first printer, you can use 'sync', it will keep the original filament set on that printer, if you don't change them

TheCableGuy99, 2023-03-02:
Sorry but this isn't working for me. I've tested with 2 out of 3 printers...

Powered on both printers.
Selected the first printer in the "Device" tab
Went to the prepare tab and selected to "Resync"
Changed each slot to my filament profiles
Selected the second printer in the "Device" tab
Went to the prepare tab and selected to "Resync"
Switch back to the first printer in the "Device" tab and the second printers filaments are showing
Clicked the "Sync" button and the colours change back to the colours for printer 1 but the profiles stay the same for printer 2.
Switched back to Printer 2 and "synced" again and the colours change back to printer 2's colours.

So only the colours are changing and not the profiles.

Also, why do we need to sync it? Why can't it just store the folaments automatically and load them when switching printers? Then if we've made changes at the printer we can re-sync an re-enter it or even better just change it in the slicer?

Thanks.

TheCableGuy99, 2023-03-02:
Sorry I meant I was clicking "Sync" not "Resync" between changes as you advised.

bambu123, 2023-03-02:
For multiple printers, there is a workaround but with a little manual work.
First, save a demo project for every printer with filaments correctly configured.
When starting to work on a printer, open the related project first, then add more models to slice.

TheCableGuy99, 2023-03-02:
Can we not see this implemented into the software to work with ease? It's a massive pain with 8+ AMS's over 3 machines. I'm not the other person struggling and having to re-enter filaments each time.

lanewei120, 2023-03-02:
> Can we not see this implemented into the software to work with ease? It's a massive pain with 8+ AMS's over 3 machines. I'm not the other person struggling and having to re-enter filaments each time.

Hello @TheCableGuy99 

in this version, we added this logic:
1. saved machine's filament list into config file
2. add the sync and resync just as described above
when click sync, we will compare the machine's filament with previous saved ones
if it doesn't change, then we won't change the preset selected before
however, as you switched printer, the presets selected have been changed
so this solution still can not satisfy your use case...

as the presets selected are stored to 3mf in our software
it is difficult to change it to the config file, and do the auto switch when you click 'sync' after using different printer's 'resync'

so to solve your case currently, how about try as follows:
create empty 3mfs with correct filaments settings for each printer(resync+select presets)
after you select different printers, firstly switch the 3mf firstly, and do the 'sync' there
and then you can do the model import and operations

we will add a final solution to solve third-party filament support issue
however, it needs lots of time and involve other teams to develop

TheCableGuy99, 2023-03-02:
Hi,

Yes it's a workaround I can use for now, I just look forward to the day it's all resolved and we can just change printers and not have to keep setting the filaments.

Thanks for your confirmation you will resolve this in the future :)