bambulab/BambuStudio #1124: [Bug] Nozzle dripping on to plate, not going in bucket, blobs etc...

TheCableGuy99, 2023-01-20:
**Describe the bug**
I'm finding that I can't start a print reliably without having to jump up and run to the other room during the pre-printing sequence because some filament hasn't been wiped properly, the nozzle hasn't been cleaned properly, filament hasn't fallen in the purge bucket or any other reason the nozzle might be dripping and end up on the plate rather than the purge bucket.

This can either be during the pre-print sequence, a lot of times it leaves a little blob (probably because it's not retracting enough) right at the end of the first layer before it goes to inspect it, or other times it should be retracting like when probing the bed height, or even it's oozing when cooling down after settling in it's final resting place after a print.

So unless we clean the nozzle manually at the end of each print (defeats the object of the cleaning feature) then it doesn't start the next print properly it might not start the next print right either (not the only reason it happens). 

Basically I'm having to remove filament from the plate on most prints as it starts the pre-printing sequence because the current method it uses isn't working as well as it should/could. I don't like putting my hand inside when it's moving about either to do this to ensure I have a clean bed.

**3mf File for This Bug**
Not a specific file, just when starting a print mainly.

**To Reproduce**
Steps to reproduce the behavior:
Start any print and observe

**Expected behavior**
I would expect no dripping/ozzing before, during, after the first layer or after a print ozzing from the nozzle. I'd like you to re-evaluate the start and end code with a view to altering the way the wiping, cleaning and retraction is done to make this work better.

Please note, this is not just a case of me needing to replace the nozzle wiper, it's happening on 3 different X1C's I own and other people I've spoken to have the same issue.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

zeroemso, 2023-01-23:
do a video and post this on yt. iam reading alot in the groups, i own also 2 of them i never had those issues and i didnt read about  that neither. 1000h so far on 2 printers.

TheCableGuy99, 2023-01-24:
Just to add to this for when you start looking into it...

I've noticed the blobs/dripping happen mainly only higher temp filaments which makes sense I suppose. I'm also not using too high a temp because I've tested all filaments I use with temp towers (and other tests to dial them in). 

I appreciate now and again you will get filament that misses the purge bucket or doesn't get wiped properly but if that can be looked at in a way to make it more reliable it would be great.

As for the dripping, maybe add some larger retraction during the pre-printing setup/tests to stop any contaminating the build plate area?

Thanks.

SaltWei, 2023-01-28:
> As for the dripping, maybe add some larger retraction during the pre-printing setup/tests to stop any contaminating the build plate area?

Response from printer firmware team because start gcode is provided by them:
We have tried 2mm retraction in start gcode and this caused high possibility to be clog.

SaltWei, 2023-01-28:
Thank you very much!
Could you share one video aboud your case and 3MF file for us to check and reproduce?

TheCableGuy99, 2023-01-29:
Hi,

Okay this video is TPU but I also replicated it on PETG, but PLA and ABS do not seem as bad. 

You will notice that the nozzle is ozzing during the scanning of the bed before printing so providing you a 3mf file is pointless because it's all in the pre-printing sequence. Why does the nozzle need to be at printing temp so it drips during scanning of the bed?

Also it leaves a trail as it finishes the calibration lines and moves to the back of the plate to perform the final purge line in the corner.

As said, I can replicate this on PETG so it's not just TPU as you also acknowledge there is an issue with. I just think a little more thought needs to go into when the nozzle is hot and not to prevent this as it's not very safe to be putting hands inside to remove bits during the startup sequence as these bits can ruin a print and the start sequence is already anything up to 10 minutes so no one wants to be stopping it mid-way through.

Here's a link to the video: https://clipchamp.com/watch/wXG3rw7n449

Thanks!

SaltWei, 2023-01-29:
> Why does the nozzle need to be at printing temp so it drips during scanning of the bed?

Response from printer team:
Waiting temp down is too slow.

TheCableGuy99, 2023-01-29:
Thanks but as you can see from the video it's causing problems. So what can be done about it? We can't be sticking out hand in during the pre-printing sequence it's dangerous. So we either leave and it ruins the first layer and possibly the print or something needs to change in the sequence.

SaltWei, 2023-01-29:
If waiting time can be accepted, user can change start gcode to wait temperature down before scanning.
[image]

TheCableGuy99, 2023-01-29:
I've no problem adding additional time but can the fans not come on to help it cool down quicker then go off again when the temp is reached?

As well as that does it retract 2mm when it's moving to the back to do the the final purge line after the calibration lines? Because as you see in the video it also leaves a little ozzing there as well. Not as much but i've seen it worse than in the video as well. Can something be done about that as well?

Thanks for your help!

SaltWei, 2023-01-29:
> but can the fans not come on to help it cool down quicker then go off again when the temp is reached?

M106 S255 is opening the part cooling fan. So maybe you can change start-gcode like this to help cooling.
[image]

TheCableGuy99, 2023-01-29:
Is this code in the correct place? It seems to cool down before doing the calibration line. Should it not be further up somewhere near:
`;===== for Textured PEI Plate , lower the nozzle as the nozzle was touching topmost of the texture when homing ==
;curr_bed_type={curr_bed_type}
{if curr_bed_type=="Textured PEI Plate"}
G29.1 Z-0.04 ; for Textured PEI Plate
{endif}

;===== draw extrinsic para cali paint =================`

SaltWei, 2023-01-29:
> As well as that does it retract 2mm when it's moving to the back to do the the final purge line after the calibration lines?

Seem to be a problem, I haven't seen retraction here.
Maybe the right gcode should be like as following
[image]

SaltWei, 2023-01-29:
> Is this code in the correct place? It seems to cool down before doing the calibration line. Should it not be further up somewhere near:

I think not, you can see ";===== draw extrinsic para cali paint =================`" here.
Should not do cooling for extrusion on bed which may cause bed adhere issue. Just my guess and not try.

Since start gcode is open to user to edit, you can edit and check whether it helps.

TheCableGuy99, 2023-01-29:
Well I have to admit I'm not good with GCODE so need a little help here. I've added the code for the retraction at the end now but it appears to be cooling AFTER it's done the scan and is at the front left of the printer.

It needs to cool BEFORE doing the LIDAR scan of the bed. It can cool while it's scanning as long as the fan is on because that makes no difference if it's stood still or scanning. But it must be set to start cooling just before it scans the bed and start re-heating immediately after it's finished.

I'm using the high-temp plate but I don't think that makes a difference as it's the same plate I took the video using.

TheCableGuy99, 2023-01-29:
Okay I've just set a large print going so I could monitor it during the scanning process to double confirm this is not cooling during the scan and it's not. Please see my notes of the pre-printing process and where I've highlighted a couple of issues:

Pre-heats bed
Homes X and Y and moves over purge bin
Heats hotend to 250° and loads filament
reduces heat to 200° and wipes nozzle
Vibrates
Raises bed
Wipes nozzle and sets nozzle to 140°
Cleans nozzle on back of plate still lowering temp to 140°
Homes X Y and Z
Scans calibration board
Moves around a lot at the front and back to the middle whilst vibrating
Heating nozzle back to 220° and scans bed at that temp <<< this is still at 220° when scanning
Moves to the middle
moves to front left and draws purge line
Drops temp to 140° and waits with fan on<<< this is where the code we entered is I think
At: 145° it starts scanning the purge line at the front of the print
Heats to 220° and moves to back of the plate to draw the final purge line before starting the print.

Thanks.

TheCableGuy99, 2023-01-29:
Correct me if i'm wrong but I believe the last line in this code might be heating up the nozzle just before scanning the bed and causing the ozzing:
;=============turn on fans to prevent PLA jamming=================
{if filament_type[initial_tool]=="PLA"}
    {if (bed_temperature[current_extruder] >45)||(bed_temperature_initial_layer[current_extruder] >45)}
    M106 P3 S180
    {elsif (bed_temperature[current_extruder] >50)||(bed_temperature_initial_layer[current_extruder] >50)}
    M106 P3 S255
    {endif};Prevent PLA from jamming
{endif}
M106 P2 S100 ; turn on big fan ,to cool down toolhead

M104 S{nozzle_temperature_initial_layer[initial_extruder]} ; set extrude temp earlier, to reduce wait time << this line
------------------------------------------------------------------------------------------------

If I am correct can this single line just be removed or does it need altering?

SaltWei, 2023-01-29:
> Heating nozzle back to 220° and scans bed at that temp <<< this is still at 220° when scanning

[image]

SaltWei, 2023-01-29:
Or ignore the step2， just delete the line in step1.

TheCableGuy99, 2023-01-29:
Okay this is MUCH better and hardly any time increased. It only adds a few seconds while it heats the nozzle after scanning and before drawing the purge line. If it stops it ozzing for the sake of 10-15 seconds it's a massive bonus in my book.

I do need to check one thing though....

After scanning the bed it moves to the front left (by the door) and starts heating up to the first layer temp (220°) before drawing the first purge (or calibration) line. However it starts just before it gets to temp, at 210° rather than 220°.

Why is this?

If it's set to start 10° below whatever the first layer temp is that is fine. But if it's set to start at 210° no matter what filament is in there then there is a problem with the way we are now heating it up. It will start to try push filament out at temps way below what some filaments need.

Can you confirm the method it's programmed to heat up to for that purge line?

Thanks.

SaltWei, 2023-01-29:
> However it starts just before it gets to temp, at 210° rather than 220°.
> 
> Why is this?

I think should only just delete the line in step1 and then it will not have this strange behaviour.

I only know the calibration line should be printed with nozzle_temperature rather than nozzle_temperation_initial to make most printing state same with calibration line.

SaltWei, 2023-01-29:
Since start gcode and prepare steps is designed and provided by printer firmware team rather than slicer developer. I recommand reporting your concern on https://forum.bambulab.com/ or BanbuHandy APP. And that will make response more timely and accurate.

I have sent this link to printer firmware team, and hope they can give more accurate response.

TheCableGuy99, 2023-01-29:
> 
> I think should only just delete the line in step1 and then it will not have this strange behaviour.
> 
I have only deleted the line and didn't move the code below like in step 2.

I don't see the point creating another thread on the forums which will just duplicate the issue. If the firmware team are alerted to this thread I'll wait for them to read up and reply.

I only need them to confirm that it will start the purge line 10° below the first layer temp each time and not at the fixed temp of 210° which is too low for some filaments.

I'd test it myself but i'm not being a guiney pig and clogging up my nozzle if it's not set right now by us removing that line.

Thanks.

SaltWei, 2023-01-30:
1 Printer will consider the temperature reached when the difference between the extruder temperature and the target is 5°C
2 The temperature displayed on screen is later than the real temperature of extruder.

TheCableGuy99, 2023-01-30:
okay thank you. 

Last question then... will these changes be implemented into future updates as they fix a problem in my book.

Otherwise i'll have to make a new profile with these changes every time there's an update just in case you changed something else?

Thanks.

tangketan, 2023-02-02:
We have fixed this issue in the the next version of firmware: the nozzle temperature is lowered down before scanning.

SaltWei, 2023-02-02:
> As well as that does it retract 2mm when it's moving to the back to do the the final purge line after the calibration lines?

This will also be fixed in start gcode to do retraction

SaltWei, 2023-02-02:
Reopen tickits for easy tracking.
Will close it when fix is included in user version.

TheCableGuy99, 2023-02-02:
okay thanks :)

matteohoeren, 2023-08-15:
Any updates on this issue?