bambulab/BambuStudio #1127: Initial layer flow

npoulos, 2023-01-20:
**Is your feature request related to a problem? Please describe.**
When using a 0.2 nozzle and the textured bed, it's hard to get the first layer down nicely as there is not enough filament being pushed through to fill the gaps in the texture. Currently the profile does a 0.1mm first layer but i've found 0.08mm works much better on flat surface beds but on textured theres just not enough flow.

**Describe the solution you'd like**
If there was ability to control first layer flow to force more material out (but not draw a wider line like in first layer line width) it would fill the small gaps in the textured plate allowing for better adherence. 

**Describe alternatives you've considered**
Cura has this feature which i've found works great for small nozzles on textured sheets.

Cosmo991119, 2023-05-23:
we're testing that.

QingZhangBambu, 2023-05-23:
we‘re testing that

SaltWei, 2023-07-11:
Included in [1.7.](https://github.com/bambulab/BambuStudio/releases/tag/v01.07.00.53).
Open develop mode and you can see more parameters including flow of initial layer.
[image]

closed 2023-07-11 by SaltWei