bambulab/BambuStudio #1128: New Extrusion Calibration doesnt show Filamentlist

Marow, 2023-01-20:
I downloaded the newest Firmware on my P1P and also Bambu Studio.

I also tried to reinstall it completely and also connected the printer completely again.

I just see this
[image]

I cannot choose any Filament, so i dont get any numbers at the bottom 3 boxes and cannot calibrate :/

KD7LRJ, 2023-01-20:
I do have filament options in my dropdown, but once I select one, I cannot change the values for the three settings. 

As I understand the text, the filament dropdown should just prepopulate those three items with default values for the selected filament type as a starting place. In my case, I chose "Generic PETG @BBL P1P" which gave me values that are close to what I'm using for Overture PETG (255C, 80C and 10.00mm3).

However, now I am stuck as well because I can't change the values manually. I can put my cursor in the boxes and highlight the numbers, but I can't change them. I want to use 8.0mm3 as my volumetric speed, but am unable to make any changes.

Running the calibration with the default numbers showed that I needed a K=0.060 instead of the default K=0.020. This seems to be helping so far (the first layers of a print I do often looks much better than before).

d3s0l3t0r, 2023-01-20:
Same here.  No filament selection and no option to enter values ​​manually

role79, 2023-01-20:
Same here. No filament selection and no option to enter values ​​manually.

linx25, 2023-01-20:
Same here - no Filament list. Have no AMS connected.

linx25, 2023-01-21:
It seems there is a Language bug - you have to switch to the english GUI
ctrl + P

Marow, 2023-01-21:
> 

True! Thats it!

d3s0l3t0r, 2023-01-22:
Yep.. Changing to english solved the problem

lanewei120, 2023-01-24:
thanks for reporting this issue

we have fixed it in local , and it will be ready in next version

currently please switch to English for calibration if needed,
and switch back, then input the k value got from calibration in English

Sorry for the inconvenience

SaltWei, 2023-03-01:
FIx is available in v1.5.0 Public Beta

closed 2023-03-01 by SaltWei