bambulab/BambuStudio #1130: BambuStudio Linux build plate not shown

ekatus, 2023-01-20:
**Describe the bug**
Successfully (!) built the BambuStudio on Ubuntu 22.04. After starting a new or opening an existing project, the canvas on the right does not show the build plate(s) with its objects, just a blank

**3mf File for This Bug**
not really applicable, for existing projects, check e.g. with armorplate.3mf

**To Reproduce**
1. Launch BambuStudio
2. Open existing Project or Launch new 
3. recognize blank build plate canvas

**Expected behavior**
The canvas like the latest Version on my mac. 

[image]
[image]

**Desktop (please complete the following information):**
 Ubuntu on Intel NUC i5
 - Version 22.04
 - 16 GB RAM

ekatus, 2023-01-21:
Hi there, thanks for your reply. I studied the instructions @ fe13ca0 but I'm afraid this is - for the time being - beyond me. I can write you a routine in S/370 assembler, but this Linux make and config is not what I really know about. Will take some time. So far I understand the problem is with wxWidgets and I need to sweet talk the build system to pick up a customized one instead of the system provided implementation which uses GLEW.

SaltWei, 2023-03-02:
Please use v1.5.0 Public Beta

closed 2023-03-02 by SaltWei