bambulab/BambuStudio #1131: [Feature Request] Please add the option to select "not connected" on the sparse infill.

BambuLabDiscord, 2023-01-20:
I am trying to print a set of the Head(amame) headphone cones and need the ability to select that the infill in not connected to perimeters. In Super Slicer this option is under "print settings" tab and under the "infill" selection on the "Sparse" properties.

littleferris, 2023-02-16:
I think the setting you are referring to is infill anchor, which indeed is not in bambustudio but is the setting needing to adjust for the headamame.

SaltWei, 2023-04-06:
> but is the setting needing to adjust for the headamame.

@QingZhangBambu Please evaluate whether this setting is necessary

millpreet, 2023-04-17:
https://github.com/SoftFever/OrcaSlicer/issues/136 same as this issue posted in orca slicer. any work around to add it into profile without putting in gui temporarily @QingZhangBambu ? I tried this in the process JSON, 
[image]
no luck.

SaltWei, 2023-07-11:
[image]
Add back in https://github.com/bambulab/BambuStudio/releases/tag/v01.07.00.53

closed 2023-07-11 by SaltWei