bambulab/BambuStudio #1132: NahimicOSD issue

shuwn, 2023-01-20:
**Describe the bug**
Since I upgraded to 1.4 or later, the following error occurred when I opened BambuStudio.

**To Reproduce**
Appears as long as restart (doesn't affect execution)

**Expected behavior**

> short-term solution
> Changing the file name from NahimicOSD.dll to NahimicOSD1.dll can avoid this problem, but this is not a correct way!

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version: 22H2

RobinWhitfield, 2023-01-22:
This sounds like expected behaviour.
It's informing you that you could have issues as the injected DLL is known to cause issues.

If it didn't inform you, you might have issues without being aware of the cause.

What exactly is the bug here?

lanewei120, 2023-01-24:
there is a known issues when using NahimicOSD with opengl on windows
and we have some customer reporting such issue before

it is found in lots of software, so we add this logic to give this hint

hisptoot, 2023-02-21:
Did some research and test yesterday. 

I think it is easy to block NahimicOSD.dll. 

Just hook ntdll!LdrLoadDll, check the module name, stop loading if it's NahimicOSD.dll .

here is the example code:
`typedef struct _UNICODE_STRING {
    USHORT Length;
    USHORT MaximumLength;
    PWSTR  Buffer;
} UNICODE_STRING, * PUNICODE_STRING;

typedef void(WINAPI* LdrLoadDllFunc) (
    IN PWCHAR               PathToFile OPTIONAL,
    IN PULONG                Flags OPTIONAL,
    IN PUNICODE_STRING      ModuleFileName,
    OUT HMODULE* ModuleHandle);

static LdrLoadDllFunc RealLdrLoadDll = NULL;
static void WINAPI LdrLoadDllHook(
    IN PWCHAR               PathToFile OPTIONAL,
    IN PULONG               Flags OPTIONAL,
    IN PUNICODE_STRING      ModuleFileName,
    OUT HMODULE* ModuleHandle)
{
    if (ModuleFileName->Buffer != NULL)
    {
        std::wstring fileName(ModuleFileName->Buffer, ModuleFileName->Length / sizeof(WCHAR));
        if (std::wstring::npos != fileName.find(L"NahimicOSD"))
        {
            return;
        }
    }

    if (RealLdrLoadDll != NULL)
    {
        RealLdrLoadDll(PathToFile, Flags, ModuleFileName, ModuleHandle);
    }
}

static void BlockNahimicOSDInject(void)
{
    LdrLoadDllFunc baseLdrLoadDll = (LdrLoadDllFunc)GetProcAddress(LoadLibraryA("ntdll.dll"), "LdrLoadDll");
    if (baseLdrLoadDll == NULL)
    {
        return;
    }

    DetourTransactionBegin();

    RealLdrLoadDll = baseLdrLoadDll;
    DetourAttach(&(PVOID&)RealLdrLoadDll, LdrLoadDllHook);

    DetourTransactionCommit();
}`

KickingKing, 2024-06-18:
> This sounds like expected behaviour. It's informing you that you could have issues as the injected DLL is known to cause issues.
> 
> If it didn't inform you, you might have issues without being aware of the cause.
> 
> What exactly is the bug here?

Try opening Orca/BBL slicer each day for like a few times. It gets really frustrating.

I do not wish it to show.
SO the BUG is that.

s0l1dx, 2024-12-26:
> Try opening Orca/BBL slicer each day for like a few times. It gets really frustrating.
>
> I do not wish it to show. SO the BUG is that.

I second @KickingKing that there should be a way to disable the warning so it does not pop-up every time I launch BambuStudio.  I have not had any issues with BambuStudio crashing because of the dll.

shuwn, 2024-12-27:
> > 這聽起來像是預期的行為。 它通知您，您可能會遇到問題，因為眾所周知，注入的DLL會導致問題。
> > 如果它沒有通知你，你可能會在不知道原因的情況下遇到問題。
> > 這裡的錯誤究竟是什麼？
> 
> 嘗試每天開啟Orca/BBL切片機幾次。 這真的讓人很沮喪。
> 
> 我不希望它出現。所以BUG就是那個。

short-term solution
Changing the file name from NahimicOSD.dll to NahimicOSD1.dll can avoid this problem, but this is not a correct way!

> Try opening Orca/BBL slicer each day for like a few times. It gets really frustrating.
>
> I do not wish it to show. SO the BUG is that.

Here are some mathods you can try.

short-term solution
Changing the file name from NahimicOSD.dll to NahimicOSD1.dll can avoid this problem.

SyntaxDreamer, 2025-01-13:
For an injection to happen, it means there must be a service running that is doing this, and probably to all applications you execute, which is normally not a good idea.

Open Windows Services, seek a service called something like Nahimic.

Example:
**Nahimic service**

Stop it. The warning should be gone when you open the software.

Or you can also disable the service entirely from starting on Windows boot as well.

eb2292, 2025-02-20:
> For an injection to happen, it means there must be a service running that is doing this, and probably to all applications you execute, which is normally not a good idea.
>
> Open Windows Services, seek [...]

This is a good _temporary_ workaround. Worked for me. 

But why does BambuStudio behave this way with this driver? I don't have a single other program with the same issue. Disabling this service shouldn't be necessary.

SyntaxDreamer, 2025-02-21:
> This is a good _temporary_ workaround. Worked for me.
>
> But why does BambuStudio behave this way with this driver? I don't have a single other program with the same issue. Disabling this service [...]

It's not BambuStudio, it's the Nahimic code being injected into every program you open. I had problems with this before, it mostly seems to affect software written in Java for some reason. 

I think there is a file setting somewhere to have programs excluded (can't remember anymore), then you don't need to turn it off, you just exclude the BambuStudio executable. It has a default list of many popular apps that are excluded by default, but I guess BambuStudio is not on the list.

BambuStudio is just warning the user about it. Which is nice instead of having issues and blaming it on the program. This driver causes issues with many applications. If you search around, there are a gazillion issues caused by it, not just BambuStudio.