bambulab/BambuStudio #1133: Bambu Studio sends GCode only for 75% to Cloud - nothing is printed.

theogott, 2023-01-20:
**Describe the bug**
A clear and concise description of what the bug is.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
Goto:
https://bambulab.com/en/my/support/tickets/US230120922001

Watch the video. I have uploaded it there because it is too large for Github.
But you need to see the recorded video, it shows the problem.

**Expected behavior**
Would be that the Object would be printed.

Instead the NEW updated Bambu Studio 1.4.2.13 takes the GCode and sends here only 75% of it to the Cloud.
The switches to the Camera view and shows THE OLD PROJECT FROM BEFORE.

PLEASE Watch the video, to see the problem. Its not so easy to explain.
https://bambulab.com/en/my/support/tickets/US230120922001

OS is: Windows 11

[Better Vented Parametric Funnel.zip](https://github.com/bambulab/BambuStudio/files/10470627/Better.Vented.Parametric.Funnel.zip)

kartoffelfighter, 2023-01-20:
Seems to be a problem with the bambu servers - i fixed it by changing the printer to LAN only mode, reconnect to Bambu Studio with the shown password and it works like a charm again.

theogott, 2023-01-21:
Work today, so Ticket can be CLOSED.

lanewei120, 2023-01-24:
yes, it is a cloud issue

sorry for the inconvenience

closed 2023-01-24 by lanewei120