bambulab/BambuStudio #1135: Frozen AMS info 1.4.2

fi0, 2023-01-21:
OS: macOS 
Bambu Studio Version: 1.4.2

Bambu studio is frozen when it tries to retrieve AMS info from printers on LAN mode.
Reverting back to 1.4.1 doesnt have this issue.

bambu123, 2023-01-28:
Can you help to supply a picture to show what was happening?

fi0, 2023-01-29:
[image]

bambu123, 2023-01-29:
We use TSL to connect the printer to the LAN network from version 1.4.2.
If your printer hasn't updated, connecting with TSL will fail by timeout, and Studio will downgrade to the non-TSL method.
The duration before the timeout is a bit long. (about 10 seconds)

Suggest updating the printer or not using LAN-Only mode.

bambu123, 2023-01-31:
@fi0 Please confirm: 
1、Is the printer in LAN-Only mode？
2、Does the connecting always wait (more than 20 seconds)?

fi0, 2023-02-07:
1. Yes
2. Yes. And the connection fails on 1.4.2. But it works on 1.4.1. I've tried clearing the settings.

fi0, 2023-02-07:
So, it seems that the fallback didnt work.

fi0, 2023-02-07:
The printer firmware is 01.03.00.00

bambu123, 2023-02-07:
Please send us the logs which can be found in the configuration folder.
The configuration folder can be opened by clicking the menu "Help > Configuration Folder'.

bambu123, 2023-03-02:
@fi0 Please update the printer to use the TLS secure connection in the LAN-Only mode.

closed 2023-03-08 by bambu123