bambulab/BambuStudio #1138: [Feature request] Wait until correct chamber temp before print

BambuLabDiscord, 2023-01-21:
Since I am printing ABS I need to set the bed to 100c and aux fan 100% for 30min or so to get the chamber temp to a good temp for printing ABS.
I wish for a tick selection and option on the filament to "wait for chamber temp" and then have it heat up bed + aux fan as a pre-print procedure.
Then I don't have to babysit the printer and press print when the chamber temp is correct.

MiscPrinter, 2023-01-21:
This would be a great feature to have. I do the same when printing materials that need the heat.

mainmind83, 2023-02-14:
It would be convenient to establish a pre-heating temperature chamber in a range of temperatures...

aequasi, 2023-05-28:
Support for M191 would still be nice, even if its an approximate temp

tylergibson, 2023-07-20:
Upvoting this - this is the only daily issue I have with my printers.  Waiting 30-45min and having to manually start the print when the chamber hits a threshold is a constant thorn in my side.