bambulab/BambuStudio #1139: Does flow calibration and Lidar really make something ?

DzzD, 2023-01-21:
I had some few cases where calibration lines was not sticking and was not properly lined and the calibration did not popup any warning and continue printing without any issue... So I wonder if it really change something.

It would be nice to have a visual information of the result of calibration lsomewhere on the GUI ?

Benderrodrigu3z, 2023-01-21:
In my opinion its a full gimmick that the Lidar code is even doing anything, I removed the house from the plate and it just kept scanning and then went and printed, no errors, plate was a Bambu High Temp sheet. The code has not been turned on properly to calibrate.

DzzD, 2023-01-21:
> I am Convinced that the Lidar code is a full gimmick

So I am ;)

RobinWhitfield, 2023-01-22:
If it fails to scan correctly it fails silently and sets the K value to 0.02. With some reading of MQTT data you can get the value that the LiDAR cal produced, and I can assure you that it does indeed change as you'd expect.

Now should it silently fail to calibrate? I'm not sure about that one. It certainly makes it harder to determine whether or not it's worked.

DzzD, 2023-01-22:
Ok, that's probably why I got 0.2 default value when I use SoftFeaver fork for the first time. 

At least a popup warning, not stopping the print but asking for, would be a nice addition.

DzzD, 2023-04-02:
(Sorry closed by error)

SaltWei, 2024-06-05:
[image]
X1C can use lader to do pa calibration, and the result can be seen when finished

closed 2024-06-05 by SaltWei