bambulab/BambuStudio #1140: [Modification & Feature Request] Changes to LAN mode to reflect a true LAN mode

TheCableGuy99, 2023-01-21:
The current "LAN Mode" is confusing and incorrectly described in my option. LAN means "Local Area Network" so when putting the printer in this mode it should be able to access via the network like when using something like Octoprint. But you can't, you can only access it in LAN mode physically.

I would like to see the following changes please:
1) Change the name of "LAN Mode" to "Standalone Mode" (or something similar) to better reflect what this mode really is.
2) Add an actual LAN mode so we can access the printer by LAN rather than the cloud which goes down from time to time. 
3) Bambu Studio should have the option to specify the IP of the printers and a subnet scanning option (or specify a different subnet range) so it can add printers (Must support multiple printers).

Thanks for your consideration.

Benderrodrigu3z, 2023-01-21:
You can FTP to the printer, that's what LAN mode is, the password is displayed when turning it on but also works with it off, the username is bblp, port 21, you should know the ip, its on the Network screen on the printer

TheCableGuy99, 2023-01-21:
Fair enough, I still think it needs some integration within Bambu Studio though in order to send files directly over the LAN and monitor/control the printer as you would normally.

TheCableGuy99, 2023-01-21:
I was given incorrect information so closing this.

closed 2023-01-21 by TheCableGuy99